LightwheelAI/leisaac #18: Can not record or play while the device is cpu

32de18, 2025-08-08:
i do the script like 

.\isaaclab.bat -p .\scripts\recoder\replay.py --task=LeIsaac-SO101-PickOrange-v0 --teleop_device=so101leader --port=COM4 --num_envs=1 --enable_cameras --dataset_file=./datasets/dataset_gpu.hdf5 --device=cpu --episode_index=0

and the isaacsim  reported errors like follows,i can not grap the orange.

2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL., FILE C:\g\134410265\physx\source\physx\src\NpFactory.cpp, LINE 810
2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX Shape failed to be created on a prim: /World/envs/env_0/Scene/Scene/outlet_room/_body_0/object/visuals!
2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL., FILE C:\g\134410265\physx\source\physx\src\NpFactory.cpp, LINE 810
2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX Shape failed to be created on a prim: /World/envs/env_0/Scene/Scene/outlet_2_room/_body_0/object/visuals!
2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL., FILE C:\g\134410265\physx\source\physx\src\NpFactory.cpp, LINE 810
2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX Shape failed to be created on a prim: /World/envs/env_0/Scene/Scene/light_switch_room/_body_0/object/visuals!
2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX error: Supplied PxGeometry is not valid. Shape creation method returns NULL., FILE C:\g\134410265\physx\source\physx\src\NpFactory.cpp, LINE 810
2025-08-08 00:33:14  [Error] [omni.physx.plugin] PhysX Shape failed to be created on a prim: /World/envs/env_0/Scene/Scene/light_switch_2_room/_body_0/object/visuals!
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/wall_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/wall_backing_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/wall_left_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/wall_left_backing_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/wall_right_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/wall_right_backing_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/floor_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/floor_backing_room/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/counter_main_main_group/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/counter_right_main_group/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/fridge_housing_main_group/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/cab_3_main_group/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/stack_1_main_group_base/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/stack_2_main_group_base/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/stack_2_main_group_2/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/stack_3_main_group_base/world_fixed_joint
2025-08-08 00:33:15  [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - cannot create a joint between static bodies, joint prim: /World/envs/env_0/Scene/Scene/stack_4_main_group_base/world_fixed_joint

32de18, 2025-08-11:
gripper failing to grasp the orange during replay.i grasp the orange success but failed at replay.the replay shows i grasp nothing but the gripper removes normally.

EverNorif, 2025-08-11:
This situation can happen because we are currently using state replay. In the next few days, we will update the replay script, and I’ll let you know when it’s ready.

EverNorif, 2025-08-12:
@32de18 We’ve updated the script. You can pull the latest code and give it a try!