LightwheelAI/leisaac #22: Exception when annotate data

SherlockUnknowEn, 2025-08-13:
arm: SO101
version: [01144d4](https://github.com/LightwheelAI/leisaac/commit/01144d44226d3efcedb05813540208e83d139c37)
OS: Ubuntu 24.04
IsaacSim: 4.5

First I run this command to record data.

```
python scripts/environments/teleoperation/teleop_se3_agent.py \
    --task=LeIsaac-SO101-PickOrange-v0 \
    --teleop_device=so101leader \
    --port=/dev/ttyUSB_leader \
    --num_envs=1 \
    --device=cuda \
    --enable_cameras \
    --record \
    --dataset_file=./datasets/dataset.hdf5
```

Then I want to annotate data, run this command

```
python scripts/mimic/annotate_demos.py \
    --task=LeIsaac-SO101-PickOrange-Mimic-v0 \
    --input_file=datasets/dataset.hdf5 \
    --output_file=./datasets/dataset_annotated.hdf5 \
    --enable_cameras \
    --device=cuda 
```

but I got a Exception:

line 324 action shape:  torch.Size([6])
line 326 task_type:  so101leader
line 327 action_tensor:  tensor([[-0.0564, -1.7363,  1.5199,  0.9739,  0.0000, -0.1343]],
       device='cuda:0')
Traceback (most recent call last):
  File "/home/fengj/projects/leisaac/scripts/mimic/annotate_demos.py", line 457, in <module>
    main()
  File "/home/fengj/projects/leisaac/scripts/mimic/annotate_demos.py", line 259, in main
    is_episode_annotated_successfully = annotate_episode_in_manual_mode(
  File "/home/fengj/projects/leisaac/scripts/mimic/annotate_demos.py", line 414, in annotate_episode_in_manual_mode
    task_success_result = replay_episode(env, episode, success_term)
  File "/home/fengj/projects/leisaac/scripts/mimic/annotate_demos.py", line 329, in replay_episode
    env.step(torch.Tensor(action_tensor))
  File "/home/fengj/projects/leisaac/source/leisaac/leisaac/enhance/envs/manager_based_rl_leisaac_mimic_env.py", line 99, in step
    return super().step(action)
  File "/home/fengj/projects/IsaacLab/source/isaaclab/isaaclab/envs/manager_based_rl_env.py", line 174, in step
    self.action_manager.process_action(action.to(self.device))
  File "/home/fengj/projects/IsaacLab/source/isaaclab/isaaclab/managers/action_manager.py", line 329, in process_action
    raise ValueError(f"Invalid action shape, expected: {self.total_action_dim}, received: {action.shape[1]}.")
ValueError: Invalid action shape, expected: 8, received: 6.

I don't understand why env need 8 action shape ?

Is it because I'm using it incorrectly ?

EverNorif, 2025-08-14:
The use of MimicGen requires the end-effector pose as a basis. For ease of use, we first convert joint-position-based actions(6-dims) into IK-based actions(8-dims), and then perform the annotation. 

We have updated the documentation and code, you can refer to [here](https://github.com/LightwheelAI/leisaac?tab=readme-ov-file#extra-feature-) and try again.

EverNorif, 2025-08-15:
It seems that you need to re-collect the data, because the action conversion process requires the recorded `ee_frame_state`, which does not exist in the old version.

SherlockUnknowEn, 2025-08-15:
> It seems that you need to re-collect the data, because the action conversion process requires the recorded `ee_frame_state`, which does not exist in the old version.

Thanks a lot. Your reply is very prompt.

Alright, I have no choice but to re-collect data again.