LightwheelAI/leisaac #26: Pi0、SmolVla

32de18, 2025-08-15:
hello,i want to know have you train pi0 or smolvla success in your isaacsim enviorments。
i have tried several times .
I record 40 episodes and trained 20K steps using lerobot and not success by now.

EverNorif, 2025-08-15:
I assume your workflow is to use LeIsaac to collect simulation data, convert it to the LeRobot dataset format, train the policy using the official LeRobot repo, and finally test it in the real environment?

32de18, 2025-08-15:
> We haven’t evaluated it because we do not yet support running Pi0 or SmolVLA inference in the LeIsaac simulation environment. In our future plans, we will try to add support and evaluate the performance.

ok.

32de18, 2025-08-15:
> We haven’t evaluated it because we do not yet support running Pi0 or SmolVLA inference in the LeIsaac simulation environment. In our future plans, we will try to add support and evaluate the performance.

i run the same env by openpi(jax official model),it sucess.i think the lerobot Pi0 is not good or it maybe needs more data or the model has problem?.

EverNorif, 2025-08-21:
Currently, leisaac supports inference in simulation via async inference. 

For example, in the orange-grasping task, I tried the SmolVLA model locally and found that it can grasp the orange, but the smoothness is not as good as the fine-tuned gr00t n1.5.

closed 2025-08-21 by EverNorif