LightwheelAI/leisaac #33: The success rate of generating trajectories is low

SherlockUnknowEn, 2025-08-26:
I collected about 15 sample data and manually annotated them (without using the `--auto` option). I used the following command to generate trajectories, but from the logs, it can be seen that the success rate is very low. Total of 119 iterations, only 2 were successful. May I ask if your experiment is like this?

Q2: I also encountered an error of dividing by 0, and the program crashed.

* version: f42fed20f87bb1dcdf2e449a0e6aa52662f83a86

* task: pick orange

* GPU: Nvidia RTX 4090

* command: 
```
python scripts/mimic/generate_dataset.py \
    --device cuda \
    --num_envs 5 \
    --generation_num_trials 1000 \
    --input_file ./datasets/annotated.hdf5 \
    --output_file ./datasets/generated_mimic.hdf5 \
    --enable_cameras
```

* log:

[image]

EverNorif, 2025-08-26:
I guess you didn’t follow the subtask order during teleoperation? In the pick-orange task, we pick orange001, then orange002, and finally orange003 in sequence. You need to strictly follow this grasping order. By the way, there is also a final step to return the so101 to the reset position.

We conducted a small test where we generated 30 new samples from 3 demonstration trajectories, and the success rate was around 35%.

SherlockUnknowEn, 2025-08-26:
> I guess you didn’t follow the subtask order during teleoperation? In the pick-orange task, we pick orange001, then orange002, and finally orange003 in sequence. You need to strictly follow this [...]

Oh， I didn't notice this detail, do these three oranges have a specific picking order? I randomly selected one for clipping.
But in other words, does this project support shuffling order to pick oranges?

EverNorif, 2025-08-26:
> Oh， I didn't notice this detail, do these three oranges have a specific picking order? I randomly selected one for clipping.哦，我没注意到这个细节，这三种橘子有特定的采摘顺序吗？我随机选择了一个进行剪辑。 But in other words, does this project support shuffling order to pick oranges?但换句话说，这个项目是否支持洗牌顺序来采摘橙子？

Unfortunately, this is not supported at the moment.

SherlockUnknowEn, 2025-08-26:
> > Oh， I didn't notice this detail, do these three oranges have a specific picking order? I randomly selected one for clipping. But in other words, does this project support shuffling order to pick oranges?
> 
> Unfortunately, this is not supported at the moment.

Is the error of dividing by 0 also caused by this?

closed 2025-09-01 by EverNorif