LightwheelAI/leisaac #35: Warp CUDA error: Failed to get driver entry point 'cuDeviceGetUuid' (CUDA error 1)

xxll88, 2025-08-30:
install isaaclab 2.2.1 with isaacsim 5.0 in windows OS 
python scripts/environments/teleoperation/teleop_se3_agent.py ^
    --task=LeIsaac-SO101-PickOrange-v0 ^
    --teleop_device=so101leader ^
    --port=COM3 ^
    --num_envs=1 ^
    --device=cuda ^
    --enable_cameras ^
    --record ^
    --dataset_file=./datasets/dataset.hdf5

error：

Warp CUDA error: Failed to get driver entry point 'cuDeviceGetUuid' (CUDA error 1)
Warp CUDA error: Function cuDeviceGetUuid_f: a suitable driver entry point was not found
Warp CUDA error 36: API call is not supported in the installed CUDA driver (in function cuda_init, C:\g\163821891\warp\native\warp.cu:277)

[Error] [omni.physx.plugin] PhysicsUSD: Parse collision

EverNorif, 2025-09-01:
This seems to be a wrap-related bug. 

I noticed that you’ve already opened an issue in the wrap project, so I’ll link it [here](https://github.com/NVIDIA/warp/issues/940) for easier tracking by others.

flsmr, 2025-10-04:
I had the same error starting leisaac with the GTX 5090 (Warp error) with the 580 driver. I can confirm that downgrading to 570.181 solved the issue and teleoperation now works for me. Here are some hints on how to downgrade:

1) If you haven't **blacklisted the nouveau drivers** yet (for me this step was necessary to get the nvidia drivers installed) you can do so by:
```bash
echo 'blacklist nouveau' | sudo tee /etc/modprobe.d/blacklist-nouveau.conf
echo 'options nouveau modeset=0' | sudo tee -a /etc/modprobe.d/blacklist-nouveau.conf

sudo update-initramfs -u
```

2) Next **remove old drivers**:
```bash
sudo apt-get purge -y 'nvidia-*'
sudo apt-get autoremove -y

sudo add-apt-repository -y ppa:graphics-drivers/ppa
sudo apt-get update
```
3) Install the 570 driver:
```bash
sudo apt-get install -y nvidia-driver-570-open
```
4) Reboot.

5) Check that the correct driver is installed:
```bash
nvidia-smi
```
```bash
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.181                Driver Version: 570.181        CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
[...]
```